DHDAXCW/NanoPi-R2S-rk3328 #29: R2S lan口网络频繁中断

emoji5, 2023-02-11:
### 1、问题描述

LAN口（内置USB3.0转千兆）每天都会出现网络中断的情况，持续两三秒

### 2、系统版本

```
主机名 | OpenWrt
型号 | FriendlyElec NanoPi R2S
架构 | ARMv8 Processor x 4 (1512MHz, 50.4°C)
固件版本 | OpenWrt Rockchip (2023-01-08) / LuCI Master (git-22.331.38573-c0c4e42)
内核版本 | 5.4.188
本地时间 | Sat Feb 11 21:12:00 2023
运行时间 | 25d 21h 25m 48s
平均负载 | 0.44, 0.62, 0.55
CPU 使用率（%） | 10%  
```
### 3、日志信息
以下是问题出现时的日志信息，19:45时间段
```
Sat Feb 11 18:29:01 2023 authpriv.info dropbear[21662]: Child connection from 10.0.0.170:20033
Sat Feb 11 18:29:02 2023 authpriv.notice dropbear[21662]: Password auth succeeded for 'root' from 10.0.0.170:20033
Sat Feb 11 19:45:42 2023 kern.warn kernel: [2231904.306368] r8152 4-1:1.0 eth1: Tx timeout
Sat Feb 11 19:45:42 2023 kern.warn kernel: [2231904.309689] r8152 4-1:1.0 eth1: Tx status -2
Sat Feb 11 19:45:42 2023 kern.warn kernel: [2231904.310162] r8152 4-1:1.0 eth1: Tx status -2
Sat Feb 11 19:45:42 2023 kern.warn kernel: [2231904.310588] r8152 4-1:1.0 eth1: Tx status -2
Sat Feb 11 19:45:42 2023 kern.warn kernel: [2231904.311032] r8152 4-1:1.0 eth1: Tx status -2
Sat Feb 11 19:45:44 2023 kern.info kernel: [2231906.490527] usb 4-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
Sat Feb 11 19:45:44 2023 kern.err kernel: [2231906.522624] r8152 4-1:1.0 eth1: Invalid ether addr 00:00:00:00:00:00
Sat Feb 11 19:45:44 2023 kern.info kernel: [2231906.523211] r8152 4-1:1.0 eth1: Random ether addr 16:59:c2:18:5a:88
Sat Feb 11 20:31:01 2023 daemon.err uhttpd[3313]: env: can't execute '/etc/init.d/fa-rk3328-pwmfan': Permission denied
Sat Feb 11 20:31:03 2023 daemon.err uhttpd[3313]: env: can't execute '/etc/init.d/fa-rk3328-pwmfan': Permission denied
```
### 4、问题出现次数
下图为通过Synology接收存储的R2S日志信息，通过筛选查找，可以看到每天都会出现多次reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd的情况
[image]

xiananliu, 2023-02-18:
我也有有同样的问题

xiananliu, 2023-02-19:
udpxy 转发的iptv 一会儿就卡

emoji5, 2023-02-24:
该问题主要发生在大流量高负载的情况下（千兆环境）

xiananliu, 2023-02-25:
求大佬解决此问题，十分困扰

closed 2023-03-24 by 97lucky