DHDAXCW/NanoPi-R2S-rk3328 #32: r2s 使用 2023.02.14 的固件，Clash 配置失败

JiandanDream, 2023-03-05:
使用 [2023.02.14 NanoPi-R2S-linux.5.4_固件下载](https://github.com/DHDAXCW/NanoPi-R2S-rk3328/releases/tag/2023.02.14-Lean1) 其中的 simple 和 beggar 版本都遇到同样问题。

OpenClash 版本为 [v0.45.87-beta](https://github.com/vernesong/OpenClash/releases/tag/v0.45.87-beta)

现象：配置文件订阅后，启动 OpenClash，会循环下载配置，一直启动不了（忘记截图了），关键是**内核日志一直为空白**

猜测是 clash 没安装后，尝试删除 openClash，但会有报错：
```sh
bash-5.1# opkg remove luci-app-openclash - 0.45.87-beta
Removing package luci-app-openclash from root...
rm: can't remove '/etc/rc.d/S99openclash': Read-only file system
rm: can't remove '/etc/rc.d/K15openclash': Read-only file system
OpenClash Already Stop!
```
ssh 到路由上去，根据 [OpenClash](https://github.com/vernesong/OpenClash/releases) 安装依赖，提示安装成功后也不行。

尝试使用 passwall 这些插件，也是无法启动。

折腾了很久，最后刷了 [2022.12.27 NanoPi-R2S-linux.5.4_固件下载](https://github.com/DHDAXCW/NanoPi-R2S-rk3328/releases/tag/2022.12.27-Lean1)  就可以了。

猜测 2023.02.14 的固件 Clash 的插件还是有些问题的。

AllureLoveHJ, 2023-03-11:
同样遇到 忘记是full还是slim遇到的

fyxw, 2023-03-12:
> 使用 [2023.02.14 NanoPi-R2S-linux.5.4_固件下载](https://github.com/DHDAXCW/NanoPi-R2S-rk3328/releases/tag/2023.02.14-Lean1) 其中的 simple 和 beggar 版本都遇到同样问题。
> 
> OpenClash 版本为 [v0.45.87-beta](https://github.com/vernesong/OpenClash/releases/tag/v0.45.87-beta)
> 
> 现象：配置文件订阅后，启动 OpenClash，会循环下载配置，一直启动不了（忘记截图了），关键是**内核日志一直为空白**
> 
> 猜测是 clash 没安装后，尝试删除 openClash，但会有报错：
> 
> ```shell
> bash-5.1# opkg remove luci-app-openclash - 0.45.87-beta
> Removing package luci-app-openclash from root...
> rm: can't remove '/etc/rc.d/S99openclash': Read-only file system
> rm: can't remove '/etc/rc.d/K15openclash': Read-only file system
> OpenClash Already Stop!
> ```
> 
> ssh 到路由上去，根据 [OpenClash](https://github.com/vernesong/OpenClash/releases) 安装依赖，提示安装成功后也不行。
> 
> 尝试使用 passwall 这些插件，也是无法启动。
> 
> 折腾了很久，最后刷了 [2022.12.27 NanoPi-R2S-linux.5.4_固件下载](https://github.com/DHDAXCW/NanoPi-R2S-rk3328/releases/tag/2022.12.27-Lean1) 就可以了。
> 
> 猜测 2023.02.14 的固件 Clash 的插件还是有些问题的。

从报错看，应该是分区被挂载成只读格式了，导致无法正常写入配置文件。
如果你是使用ext4 版本，可以尝试按照下列方法解决

1.  ssh到路由器上
```
 mount -o rw,remount /
```
2. 重新修改配置并保存
3. 菜单挂载点里，取消 **/** 的自动挂载，防止重启后重新被挂载成只读

closed 2023-07-14 by DHDAXCW