DHDAXCW/NanoPi-R2S-rk3328 #36: Question: 请问如何正确地添加 Tailscale?

xukecheng, 2023-03-13:
我通过 adyanth/openwrt-tailscale-enabler 安装后，路由器本机就没网了，其他的倒是正常的。
路由器本机没网的特征是：能够 DNS 解析，能够 PING，但是没办法下载

[image]

[image]

kljkyzsl, 2023-04-18:
tailscale有openwrt的包，直接安装就行
地址：https://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages/
安装tailscale和tailscaled

xukecheng, 2023-04-18:
> tailscale有openwrt的包，直接安装就行 地址：https://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages/ 安装tailscale和tailscaled

忘记这个 issue 还是打开状态了。。

更新说明一下，无论从哪安装，目前我发现只要同时运行了 tailscale 和 openclash，都会导致本机的网络连接出现问题，所以这个问题跟从哪下载包没有关系，但是和安装后的配置有关。
可以将以下内容，加到防火墙的自定义规则当中，并重启防火墙
```bash
iptables -I FORWARD -i tailscale0 -j ACCEPT
iptables -I FORWARD -o tailscale0 -j ACCEPT
iptables -t nat -I POSTROUTING -o tailscale0 -j MASQUERADE
```
PS：我只在自己的环境中测试通过。

closed 2023-04-18 by xukecheng